hiveeyes/arduino #18: ADS1231 library error build on arduino due

sonsatthu507, 2017-04-24:
I buid ADS1231 library on arduino due. It's error. Can you fix it?. Thank

amotl, 2017-04-24:
Dear Đặng Thái,

thanks for your notification! Did you mean you tried to compile the library at https://github.com/hiveeyes/arduino/tree/master/libraries/ADS1231? We recently made some patches to support compilation on ESP8266 and as far as we know it still works for AVR.

What kind of MCU is on the Arduino Due?

With kind regards,
Andreas.

amotl, 2017-04-24:
Hi Đặng Thái,

would you please have a look at https://github.com/hiveeyes/arduino/tree/ads1231-atmel-sam/libraries/ADS1231? We can't promise it's actually working, but would love to hear feedback from you. At least, it seems to compile properly.

With kind regards,
Andreas.

sonsatthu507, 2017-04-25:
Thank very much amotl !
I will test on ardino due. I will feedback result to you.

amotl, 2017-05-04:
Dear @sonsatthu507,

are there any news on this from your side?

With kind regards,
Andreas.

sonsatthu507, 2017-05-05:
Dear Andreas
I have designed board ads1231 comunication with atmega32u4 bootloader
arduino leonardo
I don't buy chip ads1231 in Viet Nam. I have to buy on mouser. It takes 15
days.
If you like my project. You can add friend me on facebook
https://www.facebook.com/dangthaison1992
I will test it on my board
Thank you for your attention and I will feedback result to you.

sonsatthu507, 2017-05-22:
Dear Andreas 
To day I try ads1231 on arduino due. I don't read sensorValue on arduino due. But on arduino mega and leonardo it's ok. I think due suport 3.3v. I use 5v on ads1231 may be I need to add a ic buffer 5v to 3.3v. 
What do you think about it?

amotl, 2017-05-22:
Dear Đặng Thái,

both the Arduino MEGA as well as the Arduino Leonardo are based on ATmega MCUs (AVR architecture). However, the patch was intended for supporting the ARM architecture (especially for the SAM3X8E MCU, to be explicit).

I'm sorry i can't say anything about electrical details. However, we are happy to receive further news from you.

Cheers,
Andreas.

amotl, 2017-08-08:
Dear Đặng Thái,

> I'm sorry i can't say anything about the electrical details.

You are welcome to join our forum at https://community.hiveeyes.org/ and maybe ask your question there. We have some good and kind hardware people around who might have an answer for you.

There are already two threads about the ADS1231 and ADS1234 chips you might find interesting:
- https://community.hiveeyes.org/t/ads1231-driver-library-for-avr-and-beyond/286
- https://community.hiveeyes.org/t/code-fur-ads1232-oder-ads1234/392

Cheers,
Andreas.

amotl, 2017-08-08:
As we learned from a short internet search, the current library is using Arduino library functions like `portInputRegister` and `portOutputRegister`. However, to be more compatible with other architectures than AVR, `digitalWrite` and `digitalRead` should be used, actually.

See also: https://arduino.stackexchange.com/questions/13991/adafruit-dht22-library-arduino-due-not-compiling/13992#13992

So we recommend to try the library from the beer-gauge repository and we will be happy about your feedback.

amotl, 2017-08-08:
FYI: We also referenced the issue at [Problems compiling the ADS1231 library for Arduino Due and Arduino MKR1000 boards](https://community.hiveeyes.org/t/problems-compiling-the-ads1231-library-for-arduino-due-and-arduino-mkr1000-boards/475).

amotl, 2017-11-02:
Dear Đặng Thái,

did you get a chance to test the library by Jeffrey M. Kubascik mentioned above?

With kind regards,
Andreas.

sonsatthu507, 2017-11-02:
Hi amotl I don't have test the library by @jeffkub but i read it. I had tested library by https://github.com/novemberfiveco/ADS1231-Breakout ,it is ok. I with test the library by @jeffkub in the early time

amotl, 2019-06-05:
Dear Đặng Thái,

do you have any news about this issue? Are you still working or doing research on the weighing system you have been planning?

With kind regards,
Andreas.

amotl, 2020-03-18:
Hi again,

through [1], we found the ADS123X library [2] by @HamidSaffari to be the most canonical one these days. It builds upon the HX711 library [3] by @bodge and the ADS1232 [4] by @jeffkub.

Whether this will compile on an Arduino Due is a different story, but I would like to encourage everyone giving this a try and report back to us.

With kind regards,
Andreas.

[1] https://community.hiveeyes.org/t/code-fur-ads1232-oder-ads1234/392/13
[2] https://github.com/HamidSaffari/ADS123X
[3] https://github.com/bogde/HX711
[4] https://github.com/jeffkub/beer-gauge/tree/master/firmware/beer-gauge

amotl, 2020-05-25:
Dear Đặng Thái,

we hope everything is going well on your side. While we are doing some maintenance work on the repository we are humbly closing this issue now. Please feel free to reopen when you need more assistance on that topic.

Stay safe and with kind regards,
Andreas.

closed 2020-05-25 by amotl