hiveeyes/arduino #20: Issue in Compiling Library "ADS1231.H" for Arduino MKR1000 Board

jirav1, 2017-08-07:
Hi,

I am facing issue in compilation of ADS1231 Library for Arduino MKR1000 Board. I tried to compile both library (normal and particle one) but i am facing different errors in both cases.

I found that this errors are board specific because it works well for Arduino Yun board.
Can anyone please check for Compatibility for MKR1000 Board.

amotl, 2017-08-08:
Dear @jirav1,

do you have any output of compile errors to show to us? 

With kind regards,
Andreas.

P.S.: You might also want to check another library by Jeffrey M. Kubascik, which has good code quality and which we consider using. However, we didn't have the chance to actually try it. Maybe you can find some time and give us appropriate feedback? Thanks in advance!
- https://github.com/jeffkub/beer-gauge/blob/master/firmware/beer-gauge/ADS1232.h
- https://github.com/jeffkub/beer-gauge/blob/master/firmware/beer-gauge/ADS1232.cpp

amotl, 2017-08-08:
As we learned from a short internet search, the current library is using Arduino library functions like `portInputRegister` and `portOutputRegister`. However, to be more compatible with other architectures than AVR, `digitalWrite` and `digitalRead` should be used, actually.

See also: https://arduino.stackexchange.com/questions/13991/adafruit-dht22-library-arduino-due-not-compiling/13992#13992

So we recommend to try the library from the beer-gauge repository and we will be happy about your feedback.

amotl, 2017-08-08:
FYI: We also referenced the issue at [Problems compiling the ADS1231 library for Arduino Due and Arduino MKR1000 boards](https://community.hiveeyes.org/t/problems-compiling-the-ads1231-library-for-arduino-due-and-arduino-mkr1000-boards/475).

amotl, 2017-11-02:
Dear @jirav1,

did you get a chance to test the library by @jeffkub mentioned above? We would be happy about any feedback as we are facing similar problems and maintaining the older library is a pain. If this library works, we could even bug Jeff about publishing it to PlatformIO.

With kind regards,
Andreas.

amotl, 2020-03-18:
Hi again,

through [1], we found the ADS123X library [2] by @HamidSaffari to be the most canonical one these days. It builds upon the HX711 library [3] by @bodge and the ADS1232 [4] by @jeffkub.

Whether this will compile on an Arduino MKR1000 is a different story, but I would like to encourage everyone giving this a try and report back to us.

With kind regards,
Andreas.

[1] https://community.hiveeyes.org/t/code-fur-ads1232-oder-ads1234/392/13
[2] https://github.com/HamidSaffari/ADS123X
[3] https://github.com/bogde/HX711
[4] https://github.com/jeffkub/beer-gauge/tree/master/firmware/beer-gauge

amotl, 2020-05-25:
Dear @jirav1,

we hope everything is going well on your side. While we are doing some maintenance work on the repository we are humbly closing this issue now. Please feel free to reopen when you need more assistance on that topic.

Stay safe and with kind regards,
Andreas.

closed 2020-05-25 by amotl