hiveeyes/arduino #23: Failure to clone recursively (libraries/Watterott-Arduino-Libs)

hoijui, 2019-06-04:
hello! :-)

checkout command (as in README):

    git clone --recursive https://github.com/hiveeyes/arduino

error:

    error: Server does not allow request for unadvertised object 074cae4f4eb94eae1ce83416a26dd8683104305e
    Fetched in submodule path 'libraries/Watterott-Arduino-Libs', but it did not contain 074cae4f4eb94eae1ce83416a26dd8683104305e. Direct fetching of that commit failed.

amotl, 2019-06-05:
Dear Robin,

thanks for writing in and letting us know. Sometimes we have to deal with moving targets here and elsewhere. We will see what we can do for you.

With kind regards,
Andreas.

amotl, 2020-03-18:
Dear Robin,

after checking through #25, it looks like everything seems to be in place again. So, I am deliberately closing this. Thanks again for the heads up and please feel free to reopen if you are still observing problems.

For further repository and library maintenance, I've just created #26.

With kind regards,
Andreas.

closed 2020-03-18 by amotl