hiveeyes/arduino #31: "node-esp8266-generic"

thehi1, 2021-05-02:
I am trying to compile the file "node-esp8266-generic".
Neither under Atom nor with the latest version of the Arduino IDE can the scatch be compiled.
It concerns HX711. if #define WEIGHT false, then it works.
In the meantime I have tried 3 different libraries. Unfortunately without success.
Message from Compiler:
exit status 1
no matching function for call to 'HX711::HX711(const uint8_t&, const uint8_t&)'
What could be the cause?

amotl, 2021-05-02:
Hi @thehi1,

thank you for writing in. We will probably have to update the code to use the most recent version of the HX711 library.

With kind regards,
Andreas.

cc @ClemensGruber

amotl, 2021-05-02:
Hi @thehi1,

similar to #33, #34 will bring a build environment based on PlatformIO to this program. In order to run the build, just invoke `make` in the respective program folder. Along the lines, this patch also adjusts the program to use the most recent library versions and will mitigate the hiccup you observed with the HX711.

Please let us know about the outcome.

With kind regards,
Andreas.

amotl, 2021-05-27:
Hi @thehi1,

please make sure to use the most recent version of Bogdans HX711 library and use respective code to control it, like [1]. The key is that you should use the new initialization method like
```
HX711 scale;
scale.begin(LOADCELL_DOUT_PIN, LOADCELL_SCK_PIN);
```

> I am trying to compile the file "node-esp8266-generic".

You probably used the outdated version [2]. When you will look again at [3], you will receive the modernized version. The corresponding patch to get things straight was https://github.com/hiveeyes/arduino/commit/c4da975a11.

Thank you very much for notifying us about this problem. Please let us know if this works for you.

With kind regards,
Andreas.

[1] https://github.com/bogde/HX711/blob/master/examples/HX711_basic_example/HX711_basic_example.ino
[2] https://github.com/hiveeyes/arduino/blob/27524ff41d44fcd317fa585664e72fea9f7a7421/node-esp8266-generic/node-esp8266-generic.ino
[3] https://github.com/hiveeyes/arduino/blob/master/node-esp8266-generic/node-esp8266-generic.ino

PR #34 (2021-05-02): Modernize "node-esp8266-generic"
Hi there,

in order to resolve #31, this patch brings in a build environment based on PlatformIO. In order to make the program compile, some minor adjustments had to be made to be compatible with the most recent library versions.

With kind regards,
Andreas.
closed 2021-11-09 by amotl