hiveeyes/arduino #32: node-wifi-mqtt.ino

thehi1, 2021-05-02:
I am also trying to compile the file "node-wifi-mqtt.ino". 
The following message appears when compiling:
 (I am using the Arduino IDE 1.8.13).
The following message appears when compiling:
exit status 1
'SerialDebug' was not declared in this scope
What could be the cause?

amotl, 2021-05-02:
Hi @thehi1,

thank you for writing in.

> 'SerialDebug' was not declared in this scope

Can you look on the Arduino Library Manager for a corresponding library? It is https://playground.arduino.cc/Code/SerialDebugger/, actually.

With kind regards,
Andreas.

amotl, 2021-05-02:
You can also get all of the libraries used within the programs into the `libraries` folder by invoking:
```
git submodule update --init --recursive
```

amotl, 2021-05-02:
Hi @thehi1,

#33 will now use a different SerialDebug library. Because the `SerialDebugger` by Arduino referenced above seemed very much outdated, we switched over to another library by Joao Lopes which seems to be amongst the most popular ones when searching on PlatformIO [1].

Please let us know if that actually works. At least, it compiles without errors on my machine when invoking `make`.

With kind regards,
Andreas.

[1] https://platformio.org/lib/search?query=serialdebug

PR #33 (2021-05-02): Modernize "node-wifi-mqtt"
Hi there,

in order to resolve #32 and modernize the program `node-wifi-mqtt.ino`, this patch brings in:

- Use build environment based on PlatformIO
- Use SerialDebug library by Joao Lopes

With kind regards,
Andreas.

cc @thehi1
closed 2021-05-02 by amotl