hiveeyes/arduino #58: Dissolve previous build tooling. Move on to PlatformIO.

amotl, 2023-05-27:
## About

With GH-57, we are modernizing our first-generation multi-hop ISM packet radio firmwares, effectively dissolving the previous Makefile-based build tooling, using dependency management based on Git submodules, and move on to use PlatformIO exclusively.

## Acknowledgements

We want to send big shoutouts to all the authors of the excellent build tooling which supported the programs in this repository for such a long time, sharing the same spirit that none of us wants to actually use the Arduino IDE, focusing on AVR/Espressif/Arduino HAL toolchain setup automation instead. Thank you so much!

- https://github.com/amperka/ino
- https://github.com/sudar/Arduino-Makefile
- https://github.com/ladislas/Bare-Arduino-Project
- https://github.com/plerup/makeEspArduino

/cc @einsiedlerkrebs, @amperka, @nkrkv, @sudar, @tuna-f1sh, @peplin, @rpavlik, @matthijskooijman, @ladislas, @plerup, @ivankravets, @valeros

ivankravets, 2023-05-29:
Thanks for using PlatformIO! 🙏 

We see [you use outdated EPS32 dev-platform](https://github.com/hiveeyes/arduino/blob/main/ringlabs/bienenwaage-5.0/platformio.ini#L15). Do you have any issues with the latest 6.x version https://github.com/platformio/platform-espressif32/releases ?

amotl, 2023-05-29:
Dear Ivan,

thanks for your swift response. I've been conservative in bringing in the PlatformIO configuration, and nailed the SDKs roughly to versions they had at the time the corresponding programs have been conceived. Now, upgrading the ESP32 platform SDK works well, see GH-68.

However, upgrading ESP8266 with GH-67 failed at [^1]. It is probably our fault, because we are using other outdated libraries, or such. I will investigate. 

It turns out [joaolopesf/SerialDebug](https://registry.platformio.org/libraries/joaolopesf/SerialDebug) might trip the build, as we receive similar error messages like outlined in https://github.com/JoaoLopesF/SerialDebug/issues/20 when updating to `espressif8266@^3`.

```cpp
SerialDebug.cpp:2384:46: error: reference to 'byte' is ambiguous
SerialDebug.cpp:2384:46: error: 'byte' has not been declared
SerialDebug.cpp:2486:61: error: reference to 'byte' is ambiguous
SerialDebug.cpp:5608:21: error: expected primary-expression before '(' token
```

Regarding this `SerialDebug` library, I can find those comments in the changelog:
```
 2017-04-07 Silence debug output by using SerialDebugger to reduce battery drain even more.
 2021-05-02 Modernize code by using the SerialDebug library by Joao Lopes and
            add build environment based on PlatformIO.
```

Is it sane? Do we need it for the purposes outlined there, or do you think it can be just removed?

With kind regards,
Andreas.

[^1]: https://github.com/hiveeyes/arduino/actions/runs/5113724917/jobs/9193256917?pr=67#step:8:130

P.S.: Those commands should get anyone interested to the right spot quickly.
```shell
git clone https://github.com/hiveeyes/arduino hiveeyes-arduino --branch amo/update-esp8266-sdk
cd hiveeyes-arduino/backdoor/node-wifi-mqtt
make
```

amotl, 2023-05-29:
The next spot are the firmwares based on Homie, which are still using `marvinroger/Homie@^2`, while `marvinroger/Homie@^3` is already available. /cc @Gulaschcowboy 

```
Homie/Boot/BootConfig.cpp:282:20: error: call to 'HTTPClient::begin' declared with attribute error: obsolete API, use ::begin(WiFiClient, url)
  282 |   _httpClient.begin(url);
      |   ~~~~~~~~~~~~~~~~~^~~~~
*** [.pio/build/esp07/lib5c7/Homie/Homie/Boot/BootConfig.cpp.o] Error 1
```

-- https://github.com/hiveeyes/arduino/actions/runs/5113969818/jobs/9193750333?pr=67#step:9:267

amotl, 2023-05-29:
Hi again. 4b457972e7 and 6d72ad6f6ac were needed to stay on `espressif8266@^2` on both of the occasions reported above. Other than this, all other updates succeeded on CI, and have been merged. Thanks again.

amotl, 2023-05-29:
With GH-69, I tried to update the platform SDK for AVR, and it also failed. However, after investing a bit here, I can surely say 22c6da6efa makes me happy, by being able to dissolve another [StandardCplusplus](https://github.com/hiveeyes/StandardCplusplus) library fork we needed to use the other day, because [mike-matera/ArduinoSTL](https://registry.platformio.org/libraries/mike-matera/ArduinoSTL) did not work for us back then. Excellent.

amotl, 2023-05-29:
Hi. I've created GH-71 and GH-72 to follow up on this. Closing this now. Thanks, and keep up the spirit.

closed 2023-05-29 by amotl